mjs513/FreeIMU-Updates #94: ATMega328 + LSM9DS1

wormyrocks, 2018-01-12:
Hi,
It looks like there are some provisions in the FreeIMU code to make a version compatible with the Atmega328, but it's only compatible with the Invensense devices. I'm trying to test the FreeIMU_Serial sketch with an Arduino Uno and an LSM9DS1 and while the sketch successfully compiles and uploads, it takes about 90% of flash memory and (I believe) fills up the Uno's SRAM as soon as I try to start streaming data to the GUI, leading to unacceptably slow operation. I also have access to a Leonardo, but when I compile it for the Leonardo it can't upload as it takes up 108% of available flash.

If I wanted to to use FreeIMU with an Arduino Uno and an LSM9DS1, would it be easier to modify the default FreeIMU sketch to use less memory, or to modify the FreeIMU60XX test code to support the LSM9DS1? Or am I doing something wrong elsewhere?

Thanks very much!

mjs513, 2018-01-14:
Hi wormyrock

The LSM9DS1 sensor is already supported by the library.  All you have to do is go into the freeimu.h file comment out whatever board is currently commented and uncomment the board configuration for the LSM9DS1 sensor.

wormyrocks, 2018-01-14:
Hi - 

Yes, I got the firmware to compile and upload to an Uno, but the FreeIMU-serial sketch causes what I believe is a stack-heap collision on the ATMega328, as the global variables use nearly all the SRAM. Once I connect the board and enter 'v' into the serial monitor, I usually don't get the FreeIMU descriptor response, and if I do it is several seconds later. This unreliability / latency is far too slow to stream enough data to calibrate using the GUI.

I believe the issue is a memory error because I have tested the sketch with most of the clauses in the main 'if' statement removed (retaining all the include statements) in order to free up a few more bytes of SRAM, and the 'v' call-and-response happens instantaneously. Of course, now I cannot actually use the calibrator program. Let me know if you think my diagnosis is correct, I have little experience in diagnosing embedded failure cases.

Have you tested the firmware using an ATMega328 and the LSM9DS1? If you don't have access to these parts at the time I am happy to supply any intel you might require in figuring out what's going on.

I know I could just switch to an MCU with more memory, but I have a premade PCB with a 328 and an LSM9DS1 that is designed to use as little power as possible, and the lightweight Mahony library works just fine on it. All I am trying to do is get a set of magnetometer calibration constants.

wormyrocks, 2018-01-16:
Hi -
I think I might have made some false assumptions in trying to get FreeIMU set up.

wormyrocks, 2018-01-18:
hi - I think I've worked through this issue; the problem was that I was going from an old version of cal_gui from the FreeIMU_GUI_EXE folder. I believe that one sends the 'h' command, whereas when I actually install the python libraries and run from source as I should have, everything works as expected.

closed 2018-01-18 by wormyrocks