mjs513/FreeIMU-Updates #98: Error(s) while compiling FreeIMU_serial for Arduino Due, GY-86

vishwam93, 2018-09-28:
Arduino IDE - 1.5.7
Arduino Due
Windows 10 64 bit
GY-86

I am trying to upload the FreeIMU_serial for calibration but I am flooded with errors when I do. The only changes I've made to the sketch is commenting out the header files for all sensors but the ones on the GY-86 (MPU6050 + HMC5883L + MS5611). I've also let I2Cdev.h and BaroSensor.h stay uncommented.

Arduino: 1.5.7 (Windows 8), Board: "Arduino Due (Programming Port)"

THESE ARE THE ERRORS:

In file included from f:\thesis\arduino-1.5.7\hardware\tools\gcc-arm-none-eabi-4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\type_traits:35:0,
                 from F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:7,
                 from FreeIMU_serial_Vishwam.ino:26:
f:\thesis\arduino-1.5.7\hardware\tools\gcc-arm-none-eabi-4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
In file included from FreeIMU_serial_Vishwam.ino:26:0:
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:37:15: error: 'enable_if' in namespace 'std' does not name a type
 typename std::enable_if<std::is_integral<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value ,bool>::type
               ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:37:24: error: expected unqualified-id before '<' token
 typename std::enable_if<std::is_integral<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value ,bool>::type
                        ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:41:15: error: 'enable_if' in namespace 'std' does not name a type
 typename std::enable_if<std::is_floating_point<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value, bool>::type
               ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:41:24: error: expected unqualified-id before '<' token
 typename std::enable_if<std::is_floating_point<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value, bool>::type
                        ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h: In function 'bool is_zero(T)':
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:19: error: 'is_floating_point' is not a member of 'std'
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                   ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:43: error: expected primary-expression before '>' token
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                                           ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:44: error: '::value' has not been declared
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                                            ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:55: error: 'is_base_of' is not a member of 'std'
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                                                       ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:72: error: expected primary-expression before ',' token
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                                                                        ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:49:73: error: expected primary-expression before 'float'
     static_assert(std::is_floating_point<T>::value || std::is_base_of<T,float>::value,
                                                                         ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:50:57: error: there are no arguments to 'static_assert' that depend on a template parameter, so a declaration of 'static_assert' must be available [-fpermissive]
                   "Template parameter not of type float");
                                                         ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:50:57: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h: At global scope:
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:116:36: error: expected type-specifier before 'decltype'
 auto wrap_180_cd(const T angle) -> decltype(wrap_180(angle, 100.f));
                                    ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:116:36: error: expected initializer before 'decltype'
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:130:36: error: expected type-specifier before 'decltype'
 auto wrap_360_cd(const T angle) -> decltype(wrap_360(angle, 100.f));
                                    ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:130:36: error: expected initializer before 'decltype'
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:204:55: error: expected type-specifier before 'decltype'
 static inline auto MIN(const A &one, const B &two) -> decltype(one < two ? one : two)
                                                       ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:204:55: error: expected initializer before 'decltype'
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:210:55: error: expected type-specifier before 'decltype'
 static inline auto MAX(const A &one, const B &two) -> decltype(one > two ? one : two)
                                                       ^
F:\Thesis\arduino-1.5.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:210:55: error: expected initializer before 'decltype'
[... 86 lines cut ...]
   static constexpr float Coef1 = 1.3072850288f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:142:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:143:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.4918122372f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:143:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:144:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 2.167702007e+01f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:144:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:158:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 0.6710290908f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:158:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:159:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.2523246263f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:159:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:160:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 6.881181354e+00f;
          ^
F:\Thesis\arduino-1.5.7\libraries\AP_Filter/Butter.h:160:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
FreeIMU_serial_Vishwam.ino: In function 'void loop()':
FreeIMU_serial_Vishwam.ino:170:7: error: 'cmd1' was not declared in this scope

mjs513, 2018-09-28:
Hi. My computer is out of commission right now.  Won't be able to check until next week.

One think that might resolve your problems though is to update your arduino ide to the latest version which is 1.87. Yours is alot outdated.  They have since upgraded from c++0x to c++11 which will probably get rid of those errors.

Mike

vishwam93, 2018-09-29:
Hey Mike. Thanks for the reply. I ran it on 1.8.7 and got rid of the errors. But they are new errors now

THESE ARE THE ERRORS:

Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from F:\Thesis\arduino-1.8.7\libraries\stlport-avr-STLport-master-arduino-installed/type_traits:31:0,

                 from F:\Thesis\arduino-1.8.7\libraries\AP_Math_freeimu/AP_Math_freeimu.h:5,

                 from C:\Users\VISHWA~1\AppData\Local\Temp\arduino_modified_sketch_816376\FreeIMU_serial.ino:26:

F:\Thesis\arduino-1.8.7\libraries\stlport-avr-STLport-master-arduino-installed/cstddef:102:41: error: expected unqualified-id before 'nullptr'

 /* extern */ const _STLP_STD::nullptr_t nullptr = {};

                                         ^

C:\Users\VISHWA~1\AppData\Local\Temp\arduino_modified_sketch_816376\FreeIMU_serial.ino: In function 'void loop()':

FreeIMU_serial:170:7: error: 'cmd1' was not declared in this scope

       cmd1 = '0';

       ^

Multiple libraries were found for "I2Cdev.h"
 Used: F:\Thesis\arduino-1.8.7\libraries\I2Cdev
 Not used: F:\Thesis\arduino-1.8.7\libraries\MotionDriver
exit status 1
'cmd1' was not declared in this scope

Thanks in advance!

mjs513, 2018-09-30:
Can you tell me what sketch you are running?  Because it shouldn't even be looking at stlport-avr-STLport-maste?

mjs513, 2018-10-03:
I finally got my desktop put back together and I made a few changes for it to compile correctly for your configuration.  I have updated the GitHub library accordingly.  However, here are the updated directories that were updated:
[FreeIMU.zip](https://github.com/mjs513/FreeIMU-Updates/files/2441732/FreeIMU.zip)

vishwam93, 2018-10-23:
Hey

Thanks for the reply. I got rid of all the errors and uploaded the code to Arduino Due. However, I am unable to receive any values from the GY-86's Accelerometer and Gyroscope and am getting constant stationary values from the Magnetometer. This is the result from first typing 'v' and then 'r' on the serial monitor.

`FreeIMU library by Fabio Varesano - varesano.net, FREQ:84 MHz, LIB_VERSION: DEV, IMU: FreeIMU v0.4
0,0,0,0,0,0,8199,31284,8168,0,23.88,969.08,274620,

0,0,0,0,0,0,8199,31284,8168,0,23.88,969.08,274623,

0,0,0,0,0,0,8199,31284,8168,0,23.88,969.08,274626,

0,0,0,0,0,0,8199,31284,8168,0,23.12,967.31,274633,

0,0,0,0,0,0,8199,31284,8168,0,23.12,967.31,274642,

0,0,0,0,0,0,8199,31284,8168,0,23.12,969.02,274651,

0,0,0,0,0,0,8199,31284,8168,0,23.12,969.02,274661,

0,0,0,0,0,0,8199,31284,8168,0,23.17,969.13,274670,

0,0,0,0,0,0,8199,31284,8168,0,23.17,969.13,274679,

0,0,0,0,0,0,8199,31284,8168,0,23.17,969.01,274689,

0,0,0,0,0,0,8199,31284,8168,0,23.17,969.01,274698,

0,0,0,0,0,0,8199,31284,8168,0,23.21,969.11,274707,

0,0,0,0,0,0,8199,31284,8168,0,23.21,969.11,274717,

0,0,0,0,0,0,8199,31284,8168,0,23.21,969.03,274726,

0,0,0,0,0,0,8199,31284,8168,0,23.21,969.03,274736,

0,0,0,0,0,0,8199,31284,8168,0,23.25,969.12,274745,

0,0,0,0,0,0,8199,31284,8168,0,23.25,969.12,274754,

0,0,0,0,0,0,8199,31284,8168,0,23.25,969.07,274764,

0,0,0,0,0,0,8199,31284,8168,0,23.25,969.07,274773,

0,0,0,0,0,0,8199,31284,8168,0,23.28,969.15,274782,

0,0,0,0,0,0,8199,31284,8168,0,23.28,969.15,274792,

0,0,0,0,0,0,8199,31284,8168,0,23.28,969.09,274801,

0,0,0,0,0,0,8199,31284,8168,0,23.28,969.09,274810,

0,0,0,0,0,0,8199,31284,8168,0,23.31,969.15,274820,

0,0,0,0,0,0,8199,31284,8168,0,23.31,969.15,274829,

0,0,0,0,0,0,8199,31284,8168,0,23.31,969.09,274839,

0,0,0,0,0,0,8199,31284,8168,0,23.31,969.09,274848,

0,0,0,0,0,0,8199,31284,8168,0,23.34,969.15,274857,

0,0,0,0,0,0,8199,31284,8168,0,23.34,969.15,274867,

0,0,0,0,0,0,8199,31284,8168,0,23.34,969.10,274876,

0,0,0,0,0,0,8199,31284,8168,0,23.34,969.10,274885,

0,0,0,0,0,0,8199,31284,8168,0,23.36,969.16,274895,

0,0,0,0,0,0,8199,31284,8168,0,23.36,969.16,274904,
[... 138 lines cut ...]

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.17,275559,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.17,275569,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.17,275578,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275587,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275597,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275606,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275615,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275625,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275634,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275643,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.18,275653,

0,0,0,0,0,0,8199,31284,8168,0,23.51,969.16,275662,
`