mjs513/FreeIMU-Updates #103: Compiling FreeIMU_serial_2 Error

abdalrahman9, 2019-03-20:
Hello there,

The problem that I am having is compiling the FreeIMU_serial_2 which is suppose to be an edited version of the first version so that it can be compatibile with the new arduino IDEs. However, when I am trying to compile it under Arduino 1.8.7 or 1.8.8, I recieve the following error:

```
In file included from C:\Users\FMC\Documents\Arduino\libraries\stlport-avr-STLport-master-arduino-installed/type_traits:31:0,

                 from C:\Users\FMC\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:5,

                 from C:\Users\FMC\Documents\Arduino\FreeIMU_serial_2\FreeIMU_serial_2.ino:27:

C:\Users\FMC\Documents\Arduino\libraries\stlport-avr-STLport-master-arduino-installed/cstddef:102:41: error: expected unqualified-id before 'nullptr'

 /* extern */ const _STLP_STD::nullptr_t nullptr = {};

`
```

The person that posted issue #98 had a similar problem but you had given him the files and said that it should work but it doesn't. From what I understand is that it shouldn't be even looking at the stlport library but it is. I am trying to compile it on a LSM9DS1 through an Arduino nano. Thank you for your help.

mayankmeh, 2019-07-13:
i am also having the same problem, did you got it working?

ShubhamBhatt, 2022-01-30:
Me too any solution yet

mjs513, 2022-01-31:
You can try commenting out that line and it the file cstddef.  Unfortunately I am unable to test if it works.  Nullptr should already be defined for Arduino.