mjs513/FreeIMU-Updates #106: Free IMU library not working.

ShubhamBhatt, 2022-01-30:
Just downloaded the library. I am not able to compile it for Arduino Uno. I am using the latest Arduino IDE. Not sure if this library is still been used. If anyone out there is still using it please help.
[image]

mjs513, 2022-01-31:
See comment here:  https://github.com/mjs513/FreeIMU-Updates/issues/103#issuecomment-1025186261

However, the library won't fit on an Arduino Uno you would have to use a Mega or something with a bit more memory

ShubhamBhatt, 2022-01-31:
Thanks for the quick reply. Unfortunately commenting on the line gave more errors. Is there any tutorial for freeimu to be used with mpu6050 ?? maybe I missed something? I am not able to find any guide on how to use this library.

ShubhamBhatt, 2022-01-31:
I tried commenting most of the libraries. some places you have mentioned MPU60XX these XX need to be replace by actual library??
[image]
I get this if I make changes like above
[image]

mjs513, 2022-01-31:
Not sure what the example you are using, but I just compiled for the Mega using the Yaw_pitch_roll example and it compiled with out errors.  This is what I have for headers to be included:

```
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <AP_Baro_MS5611.h>  //Uncomment for APM2.5
```
Make sure you include the I2Cdev library when using the MPU6050